ArduPilot/pymavlink #961: Message signature verification in the python library does not properly check the timstamp.

pv42, 2024-07-31:
The function `check_signature(..) -> bool` in `mavutil.py` generated from `mavgen_python.py` https://github.com/ArduPilot/pymavlink/blob/e192ad8114f203220f404f37f971d6359dd5e3d2/generator/mavgen_python.py#L963 does not increase the value of `self.signing.stream_timestamps[stream_key]` after initially setting it on the first message received per logical stream.
The documentation says that a packet should be rejected if  

> Timestamp is older than the previous packet from the same logical stream - where a logical stream is defined as the sequence of MAVLink packets with the same (SystemID, ComponentID, LinkID) tuple.  

but since the value is never updated this comparison is only done against the first packet of the stream.

tridge, 2024-08-27:
@pv42 I think you are right and this needs to be fixed. Would you like to do a PR yourself? or shall I do it?

pv42, 2024-08-27:
Hello @tridge, thanks for the response.
I just noticed the issue when comparing the c implementation, this python implementation and the documentation. While I was reasonably confident that there is something wrong with this implementation I am not sufficiently confident to write the security critical code of a codebase I am not familiar with.

PR #1087 (2025-07-10): mavgen_python: fixed signature timestamp update
ensure we update the signature timestamp on each packet
fixes #961
commit 1db7e7ed (2025-07-15): mavgen_python: fixed signature timestamp update

ensure we update the signature timestamp on each packet
fixes #961
closed 2025-07-10 by tridge