ArduPilot/pymavlink #963: mavplayback fails playing back .BIN file

swilliam, 2024-08-09:
[mavplayback.py](tools/mavplayback.py) playing back a .BIN log file

```
$ mavplayback.py 00000038.BIN
Traceback (most recent call last):
  File "~/bin/mavplayback.py", line 256, in <module>
    app=App(filename)
  File "~/bin/mavplayback.py", line 104, in __init__
    self.next_message()
  File "~/bin/mavplayback.py", line 199, in next_message
    pos = float(self.mlog.f.tell()) / self.filesize
AttributeError: 'DFReader_binary' object has no attribute 'f'
```
[mavplayback.py] (tools/mavplayback.py) uses DFReader_binary class from [DFReader.py](pymavlink/DFReader.py) to load .BIN file, but that class has no ```f``` attribute instead it uses ```filehandle```

playback of ```.tlog``` files works

gamoreno, 2024-08-30:
I have the same problem. I replaced `f` with `filehandle` and it doesn't throw the error, but it seems to be sending invalid mavlink messages in the out port.

peterbarker, 2024-10-02:
mavplayback only works with .tlog files - it only *looks* for mavlink messages, doesn't attempt to extract data from any other log type.

We should probably check the type returned by the connection and give a reasonable error message: https://github.com/ArduPilot/pymavlink/pull/985

closed 2024-10-02 by peterbarker