ArduPilot/pymavlink #972: problems with vx and vy amendments

EVNN304, 2024-08-23:
Hello. I have a question regarding Mavlink and cuav v 5. Using a Python script, I am trying to send a speed correction to cuav v 5 and see the application of vx and vy corrections in the Mission Planner program. The connection is established with the controller via the script and the Mission Planner program itself sees the controller, but the corrections are not applied. Tell me what I am doing wrong and how this can be fixed. I will provide the code below

from pymavlink import mavutil
import time
import pymavlink
master = mavutil.mavlink_connection('COM4', baud=9600)
master.wait_heartbeat()

target_system = master.target_system  
target_component = master.target_component  

print("Heartbeat from system (system_id: {}, component_id: {})".format(master.target_system, master.target_component))

vx_adjustment = 2.5  
vy_adjustment = 1.5  

def send_speed_adjustments(vx, vy):

    master.mav.set_position_target_local_ned_send(
        0,  # time_boot_ms (not used)
        master.target_system,  # target_system
        master.target_component,  # target_component
        mavutil.mavlink.MAV_FRAME_LOCAL_NED,  # frame
        0b0000111111111111,  #
        0, 0, 0,  # x, y, z 
        vx, vy, 0,  # vx, vy, vz 
        0, 0, 0,  # x_acc, y_acc, z_acc 
        0, 0  # yaw, yaw_rate
    )
    print(f'Sent speed adjustments: vx = {vx}, vy = {vy}')

try:
    while True:
        send_speed_adjustments(vx_adjustment, vy_adjustment)
        time.sleep(1)  
except KeyboardInterrupt:
    print("Script terminated by user.")

[image]
[image]

EVNN304, 2024-08-23:
from pymavlink import mavutil
import time
import pymavlink
master = mavutil.mavlink_connection('COM4', baud=9600)
master.wait_heartbeat()

target_system = master.target_system  
target_component = master.target_component  

print("Heartbeat from system (system_id: {}, component_id: {})".format(master.target_system, master.target_component))

vx_adjustment = 2.5  
vy_adjustment = 1.5  

def send_speed_adjustments(vx, vy):

    master.mav.set_position_target_local_ned_send(
        0,  # time_boot_ms (not used)
        master.target_system,  # target_system
        master.target_component,  # target_component
        mavutil.mavlink.MAV_FRAME_LOCAL_NED,  # frame
        0b0000111111111111,  #
        0, 0, 0,  # x, y, z 
        vx, vy, 0,  # vx, vy, vz 
        0, 0, 0,  # x_acc, y_acc, z_acc 
        0, 0  # yaw, yaw_rate
    )
    print(f'Sent speed adjustments: vx = {vx}, vy = {vy}')

try:
    while True:
        send_speed_adjustments(vx_adjustment, vy_adjustment)
        time.sleep(1)  
except KeyboardInterrupt:
    print("Script terminated by user.")

rmackay9, 2024-08-23:
Hi @EVNN304,

The AP wiki has [this section detailing how to control a copter using mavlink](https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html) and it even includes commands that can be copy-pasted into MAVProxy's console.

From the picture above it looks like the vehicle has no position estimate so then the issue is the vehicle needs an alternative [non-GPS sensor](https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html).  I'm afraid that the IMUs available in today's autopilots are not the same level of quality that are found on submarines and space craft that allow them to estimate their position without any other external sensors.

I might close this issue if that's OK but we can re-open if necessary

closed 2024-08-23 by rmackay9

EVNN304, 2024-08-25:
Hi, thanks for the answer) But the question still remains. At the moment I want to understand whether I have correctly implemented the python script for sending vx and vy speed corrections? If I have implemented it correctly, then the main problem is that the cuav flight controller does not see the gps signal and because of this does not accept the received speed corrections? I apologize in advance for the strange questions, since I have been working on this topic for about a week)

rmackay9, 2024-08-26:
@EVNN304, separating the problem is probably the best way forward.  So for example test the script in the simulator to be sure it works there.  Then test the autopilot using by copy-pasting the commands from the wiki page linked above.