ArduPilot/pymavlink #976: Please straighten out your supported Python versions and dependencies

sarnold, 2024-09-13:
I just came back to this after several years and there seems to be *zero progress* in terms of cleaning up Python version support or packaging.  Meaning `setup.py` *still requires future pkg* even though several other places claim only py36 or later (and several variants thereof).  This is getting ridiculous trying to maintain a package that won't let go of out-dated dependencies.  The world moves on and even bionic has been declared lifeless.  Please lurch forward into at least the early 21st century...

peterbarker, 2024-09-15:
> I just came back to this after several years and there seems to be _zero progress_ in terms of cleaning up Python version support or packaging. Meaning `setup.py` _still requires future pkg_ even though several other places claim only py36 or later 

We're not going out of our way to support Python2 any more - if it breaks it breaks.

But we are also not going out of our way to kill it right now in pymavlink.  That day may come - and it will be a good day!

> (and several variants thereof). This is getting ridiculous trying to maintain a package that won't let go of out-dated 

I think we may be unaware of problems that our laggardly behaviour may be causing other people here.

Are you attempting to create system packages or the like?

> dependencies. The world moves on and even bionic has been declared lifeless

.... python3 *is* a possibility on Bionic.

> Please lurch forward into at least the early 21st century...

Wait, who *doesn't* like steam trains?

bobpaul, 2025-01-13:
Am I missing something? I don't see any reference to `future` in the *.py files in the project. I just built the package with python13 and without installing `future`. I haven't done extensive testing (so maybe something is broken), but I was able to run the [sender/receiver examples from mavlink-routerd](https://github.com/mavlink-router/mavlink-router?tab=readme-ov-file#sender--receiver)

Is future only required for python2?

peterbarker, 2025-01-29:
Partially closed by https://github.com/ArduPilot/pymavlink/pull/1003 which officially says we're no longer doing Python2 and happy to actively remove support code.

peterbarker, 2025-01-29:
> Derp, I can be kinda dumb sometimes. I had the mavlink project checked out (with pymavlink as a submodule) and I was doing `git grep` from within `mavlink` instead of in `mavlink/pymavlink` :-/
> 
> Attached is a patch that I think should maintain python2 compatibility without requiring the future module in python 3.11+. I understand `sorted(mydict.items())` is a bit slower on python2.7 than using `iteritems()` from future, but it should still return the same result.
> 
> [remove_future.patch.txt](https://github.com/user-attachments/files/18403029/remove_future.patch.txt).

Could you create a PR with your fixes in it, please?  Happy to walk you through that if you need it - generally knowing a github workflow is a useful life skill :-)

bobpaul, 2025-02-04:
Sure, though I would probably do things a bit differently if if's OK to _actively_ remove support. PR #1000 already gets one of the files.

peterbarker, 2025-04-22:
> Sure, though I would probably do things a bit differently if if's OK to _actively_ remove support. PR [#1000](https://github.com/ArduPilot/pymavlink/pull/1000) already gets one of the files.

I've merged https://github.com/ArduPilot/pymavlink/pull/1025 which *actively* removed Python2 support :-)

rmackay9, 2025-04-22:
@peterbarker maybe we can close this one then?

peterbarker, 2025-05-08:
> [@peterbarker](https://github.com/peterbarker) maybe we can close this one then?

Need to investigate import of future module first, I think.

amilcarlucas, 2025-12-30:
#1085 got merged. I think this one can be closed once @sarnold  confirms that whatever problem he was having is now solved.