ArduPilot/pymavlink #981: AttributeError al usar MAV_CMD_DO_ORBIT en pymavlink: "No attribute 'MAV_CMD_DO_ORBIT'"

S4klas, 2024-09-26:
Estoy intentando ejecutar un comando de órbita en mi código utilizando pymavlink, pero obtengo un error AttributeError al intentar acceder a MAV_CMD_DO_ORBIT. El error que se muestra es el siguiente:  

Traceback (most recent call last):
  File "c:/Users/labc9/Desktop/drone/main2.py", line 394, in set_orbit2
    mavutil.mavlink.MAV_CMD_DO_ORBIT,  # Comando de órbita
AttributeError: module 'pymavlink.dialects.v20.development' has no attribute 'MAV_CMD_DO_ORBIT'

He intentado cambiar los dialectos importados, probando con:
from pymavlink.dialects.v20 import ardupilotmega as mavlink2
from pymavlink.dialects.v10 import ardupilotmega as mavlink1
Sin embargo, en ambos casos sigo obteniendo el mismo error. Esto indica que el comando MAV_CMD_DO_ORBIT no parece estar disponible en los dialectos ardupilotmega de las versiones de MAVLink 1 y MAVLink 2.
Versión de pymavlink:   Name: pymavlink
Version: 2.4.41
Summary: Python MAVLink code
Home-page: https://github.com/ArduPilot/pymavlink/
Author:
Author-email:
License: LGPLv3
Location: c:\python38\lib\site-packages
Requires: future, lxml
Required-by: dronekit, MAVProxy

hadif1999, 2025-04-06:
I have the same issue

peterbarker, 2025-04-06:
Not a problem with pymavlink, just the message set it is distributed with.  Closing this one.

closed 2025-04-06 by peterbarker