ArduPilot/pymavlink #1133: Error loading generated WLua Wireshark plugin

yonatan8070, 2025-09-29:
I'm generating a MAVLink dissector for Wireshark from my custom dialect, which includes `common.xml`. In Wireshark 4.4.8, I'm now getting this error on startup from my dialect:

```Lua: Error during loading:
/home/yonatan/.local/lib/wireshark/plugins/MY_DISSECTOR.lua:6915: bad argument #5 to 'new' (ProtoField_new: Base must be either base.DEC or base.UNIT_STRING)
stack traceback:
	[C]: in field 'new'
	/home/yonatan/.local/lib/wireshark/plugins/MY_DISSECTOR.lua:6915: in main chunk
	[C]: in function 'require'
```

This error originates from the `CAMERA_IMAGE_CAPTURED` message, for which the generated output looks like this, I added a comment on the problematic line:

```lua
f.CAMERA_IMAGE_CAPTURED_time_boot_ms = ProtoField.new("time_boot_ms (uint32_t) [ms]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_time_boot_ms", ftypes.UINT32, nil)
f.CAMERA_IMAGE_CAPTURED_time_utc = ProtoField.new("time_utc (uint64_t) [us]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_time_utc", ftypes.UINT64, nil)
f.CAMERA_IMAGE_CAPTURED_camera_id = ProtoField.new("camera_id (uint8_t)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_camera_id", ftypes.UINT8, nil)
f.CAMERA_IMAGE_CAPTURED_lat = ProtoField.new("lat (int32_t) [degE7]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_lat", ftypes.INT32, nil)
f.CAMERA_IMAGE_CAPTURED_lon = ProtoField.new("lon (int32_t) [degE7]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_lon", ftypes.INT32, nil)
f.CAMERA_IMAGE_CAPTURED_alt = ProtoField.new("alt (int32_t) [mm]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_alt", ftypes.INT32, nil)
f.CAMERA_IMAGE_CAPTURED_relative_alt = ProtoField.new("relative_alt (int32_t) [mm]", "mavlink_proto.CAMERA_IMAGE_CAPTURED_relative_alt", ftypes.INT32, nil)
f.CAMERA_IMAGE_CAPTURED_q_0 = ProtoField.new("q[0] (float)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_q_0", ftypes.FLOAT, nil)
f.CAMERA_IMAGE_CAPTURED_q_1 = ProtoField.new("q[1] (float)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_q_1", ftypes.FLOAT, nil)
f.CAMERA_IMAGE_CAPTURED_q_2 = ProtoField.new("q[2] (float)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_q_2", ftypes.FLOAT, nil)
f.CAMERA_IMAGE_CAPTURED_q_3 = ProtoField.new("q[3] (float)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_q_3", ftypes.FLOAT, nil)
f.CAMERA_IMAGE_CAPTURED_image_index = ProtoField.new("image_index (int32_t)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_image_index", ftypes.INT32, nil)
f.CAMERA_IMAGE_CAPTURED_capture_result = ProtoField.new("capture_result (MAV_BOOL)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_capture_result", ftypes.INT8, nil, base.HEX_DEC) -- Line 6915
f.CAMERA_IMAGE_CAPTURED_capture_result_flagMAV_BOOL_TRUE = ProtoField.bool("mavlink_proto.CAMERA_IMAGE_CAPTURED_capture_result.MAV_BOOL_TRUE", "MAV_BOOL_TRUE", 4, nil, 1)
f.CAMERA_IMAGE_CAPTURED_file_url = ProtoField.new("file_url (char)", "mavlink_proto.CAMERA_IMAGE_CAPTURED_file_url", ftypes.STRING, nil)
```

The Mavgen command I'm using is:
```shell
python3 -m pymavlink.tools.mavgen \
            --lang=WLua \
            --wire-protocol=2.0 \
            --output="$HOME"/.local/lib/wireshark/plugins/MY_DISSECTOR \
            MY_DIALECT_XML
```

peterbarker, 2025-09-29:
That looks like it might be shirty about trying to treat a MAV_BOOL as a hexadecimal number.

I think this might be the MAV_BOOL enumeration stuff which has killed it, @IamPete1 .

I've created a [PR](https://github.com/ArduPilot/pymavlink/pull/1134) which may fix ths.  Can you test that PR, @yonatan8070 ?

yonatan8070, 2025-09-30:
Looks like it's working!
Wireshark starts up with no errors after regenerating the dialect